canopen-python/canopen #466: Missing datatype 0x16 -> UNSIGNED12

prunith, 2024-06-19:
Hi,
Could you implement USIGNED12 bits datatype ? Hex code is 0x16.

Thierry

sveinse, 2024-06-19:
I'm not familiar with UNSIGNED12 in canopen. Datatype 0x16 is according to CiA 301 UNSIGNED24. Where is UNSIGNED12 used?

prunith, 2024-06-19:
Hi @sveinse you are right i made a mistake UNSIGNED24 is missing UNSIGNED12 does not exist so far.

prunith, 2024-06-19:
I dig into code to see how it can be implemented, unfortunatly canopen use struct library that not support 24bit data. Perhaps with this work arround it could work unpack('<l', b'\x57\x15\x00' + b'\x00')[0] # unsigned littleendian

prunith, 2024-06-19:
I see also that pull request already submited to correct that #267 unfortunately it was mixed with other change and not implemented.

prunith, 2024-06-19:
Tanks @sveinse @acolomb i m just late from one week , i have not noticed new release. Anyway i confirm it is fixed with 2.3.0.
Unfortunatly i have noticed some issue with latest python_can 4.4.0 library. I ll open a new ticket if needed.

closed 2024-06-19 by sveinse

acolomb, 2024-06-19:
What issues specifically? The canopen 2.3.0 release should work around an incompatible change in python-can 4.4.0 already.  And for the next python-can release, the incompatibility is fixed again on their end already, just not released.