canopen-python/canopen #470: canopen not works with python 3.12

prunith, 2024-06-21:
With python 3.12 i got error:
  File "C:\Users\100034799\AppData\Local\Programs\Python\Python312\Lib\site-packages\canopen\objectdictionary\__init__.py", line 281, in <module>
    class ODVariable:
  File "C:\Users\100034799\AppData\Local\Programs\Python\Python312\Lib\site-packages\canopen\objectdictionary\__init__.py", line 290, in ODVariable
    INTEGER24: IntegerN(24),
TypeError: Struct() argument 1 must be a str or bytes object, not int

acolomb, 2024-06-21:
That is strange indeed. We did see this error on the builder for the documentation, but I could not reproduce locally. A fix is already committed on the master branch, you might want to try installing that.

If you can confirm this is fixed on master, we will make a new point release soon.

sveinse, 2024-06-21:
This is a strange error. I just retested with a fresh python 3.12.2 in virtual environment with the current master and I am not able to reproduce the error. Which version of canopen do you use?

acolomb, 2024-06-21:
Well, current master does include a fix. @prunith most likely tested with the latest PyPI release? That does not have the fix.

prunith, 2024-06-21:
@acolomb you are right i use latest release 2.3.0 installed with pip and not the master.

sveinse, 2024-06-21:
I tried running a fresh 24.04 ubuntu docker container and installed system python3 (3.12.3) and made a venv and installed canopen from pypi in it. Still no failure.

There is _something_ in your setup which I'm curious what triggers this.

prunith, 2024-06-21:
In my case i instelled IDLE python 12 for windows and installed package with pip command

prunith, 2024-06-21:
Yes but same issue occures with pycharm and venv based on installed python 12

ayavilevich, 2024-06-21:
fyi, I had this issue with Python 3.12 but when I upgraded to 3.12.4 it went away.

erlend-aasland, 2024-07-09:
ISTM this can be closed as not-a-bug.

closed 2024-07-09 by friederschueler

ipfilip, 2024-10-14:
For me upgrading from python 3.12.0 to 3.12.7 also fixed it