canopen-python/canopen #487: Consider bumping the `python-can` minimum required version to 4.0

erlend-aasland, 2024-07-02:
I tried running the test-suite with `python-can` 3.7, but it failed. Starting with `python-can` 4.0, it succeeds. Consider bumping the version requirement in pyproject.toml.

sveinse, 2024-07-03:
Is there a minimum python requirement in `python-can` 4.0? If there is, we might need to follow suit on minimum python version. Alternatively we must make canopen work with older python-can.

erlend-aasland, 2024-07-03:
`python-can` current `main`[^1] requires Python 3.8 or newer. That's the same Python requirement as `canopen`.

`python-can` v4.2.0 requires Python 3.7. Older versions of `python-can` has no specific requirement encoded in the `pyproject.toml` file.

IOW, the Python requirement is not a blocker.

[^1]: HEAD at `174d5600aaf3377bd44f1eab9cb53ae066d606ca`

erlend-aasland, 2024-07-03:
Seems like `python-can` v4.3.0 started requiring Python 3.8. Since that harmonises with the `canopen` Python requirement, it may make sense to bump the `python-can` requirement to 4.3.0.

acolomb, 2024-07-03:
> I tried running the test-suite with `python-can` 3.7, but it failed. Starting with `python-can` 4.0, it succeeds. Consider bumping the version requirement in pyproject.toml.

Where did you get python-can version 3.7? https://pypi.org/project/python-can/#history does not show such a version.  I'm interested in the actual failure mode, maybe we can then work around it.  Do you still have logs from your trial?

erlend-aasland, 2024-07-03:
> Where did you get python-can version 3.7? https://pypi.org/project/python-can/#history does not show such a version. I'm interested in the actual failure mode, maybe we can then work around it. Do you still have logs from your trial?

That should've been version 3.0, not 3.7; sorry 'bout that.

```sh
$ python3 -m pip install python-can==3
[...]
Successfully installed python-can-3.0.0 typing-3.7.4.3
```

Nonetheless, the test failure is:

```sh
$ python -m unittest discover test
[...]
======================================================================
FAIL: test_send_perodic (test_network.TestNetwork.test_send_perodic)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/Users/erlend.aasland/src/canopen.git/test/test_network.py", line 65, in test_send_perodic
    self.assertTrue(9 <= bus.queue.qsize() <= 13)
    ~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
AssertionError: False is not true

----------------------------------------------------------------------
Ran 125 tests in 12.323s

FAILED (failures=1)
$ python -V
Python 3.13.0b3
```

erlend-aasland, 2024-07-03:
An argument for bumping the `python-can` requirement, is that the *bustype* parameter of `can.Bus` is deprecated and scheduled for **removal in python-can v5.0.0**:

> DeprecationWarning: The 'bustype' argument is deprecated since python-can v4.2.0, and scheduled for removal in python-can v5.0.0. Use 'interface' instead.

FTR: I don't know when v5.0.0 is scheduled to be released.

acolomb, 2024-07-03:
Hm I have no clue what that error is about.  Does it work with a slightly more recent version, such as 3.3.4 from the previous major release line?  If so, I'd prefer going slow and bumping to 3.3 for example.  Distributions are slow to adapt and version 4.0.0 is only two years old now.

As for the `bustype`, our usage should be adapted for the new `interface` name of course.  But it seems mainly an issue in tests and documentation.  To ensure we don't get hit by the 5.0.0 removal, simply restrict the dependency to `<5.0.0` in our TOML?

acolomb, 2024-07-03:
Note also that my local tests failed completely because python-can 3.3.4 requires pkg_resources which is no longer in Python 3.12.

acolomb, 2024-07-03:
Actually I just successfully ran the tests with python-can==3.0.0.  It required installing pytest and setuptools within the venv, but then worked fine on Python 3.12.

erlend-aasland, 2024-07-03:
> As for the `bustype`, our usage should be adapted for the new `interface` name of course. But it seems mainly an issue in tests and documentation. To ensure we don't get hit by the 5.0.0 removal, simply restrict the dependency to `<5.0.0` in our TOML until it's known to work?

The *interface* parameter name was introduced in 4.0.0 IIRC, so if you're changing that, you're implicitly bumping the version requirement. Amending the version requirement to `~=4.0` should keep any upcoming major release away.

> Actually I just successfully ran the tests with python-can==3.0.0. It required installing pytest and setuptools within the venv, but then worked fine on Python 3.12.

I didn't try with pytest, nor 3.12. I'll give it a go.

acolomb, 2024-07-03:
> The _interface_ parameter name was introduced in 4.0.0 IIRC

No, it was already in 3.0.0 at least.

erlend-aasland, 2024-07-03:
Indeed, it was!

erlend-aasland, 2024-07-03:
Resolved with #489. It seems the other failures were related to my venv.

closed 2024-07-03 by erlend-aasland