canopen-python/canopen #488: `PdoBase.export()` is broken bco. broken dependencies

erlend-aasland, 2024-07-03:
`PdoBase.export()` depends on the third-party module `canmatrix`, which was removed from the `canopen` requirements with commit c46228f9cf2d2661166d68e7175f3e8b99064194, three years ago. This implies it has been de facto broken since that commit, unless the user accidentally had a compatible `canmatrix` library installed.

There's a couple of ways to mitigate this:

1. add back the `canmatrix` dependency, fix `.export()` (if it needs fixing), and add unit tests in order to prevent future regressions like this
2. acknowledge that the API is de facto broken and purge it from `canopen`

FWIW, my personal preference would be 2).

Discovered while working on increasing PDO coverage to 100%.

erlend-aasland, 2024-07-03:
I've added PoC-branches for each proposed solution. Happy to open a PR for whatever variant you prefer.

- 1) fix: https://github.com/erlend-aasland/canopen/commit/75fa0f54ed0f436d7d8135d4a0d98a4d5eee7877
- 2) purge: https://github.com/erlend-aasland/canopen/commit/976c6d75aa88b34e69d18896788a11c1cdd04a00

erlend-aasland, 2024-07-03:
There's also a third way to mitigate this: _conditionally_ add `PdoBase.export`, depending on if `canmatrix` is installed or not. This can be trivially added on top of https://github.com/erlend-aasland/canopen/commit/69444dbf0db231dc861410960f4a1803ce783564.

acolomb, 2024-07-03:
I prefer the third solution.  It seems this is the only place where the dependency is used, so forcing all users to install it is too much.  But the mapping export function itself is surely useful.

Can we make sure that the `canmatrix` package is installed when running tests?

erlend-aasland, 2024-07-03:
Yep, the third solution is probably easiest on users. I'll create a PR shortly.

PR #493 (2024-07-09): Add back PdoBase.export() dependency (canmatrix)
The canmatrix optional dependency was removed on Oct 10, 2021 with commit
c46228f. It is now added back as an optional dependency, using the same
name as previously: db_export.

To install the dependency:

$ python3 -m pip install 'canopen[db_export]'

Resolves #488
commit 6bd39ba8 (2024-07-09): Add back PdoBase.export() dependency (canmatrix) (#493)

The canmatrix optional dependency was removed on Oct 10, 2021 with commit
c46228f. It is now added back as an optional dependency, using the same
name as previously: db_export.

To install the dependency:

$ python3 -m pip install 'canopen[db_export]'

Resolves #488
closed 2024-07-09 by acolomb