canopen-python/canopen #496: examples/simple_ds402_node.py uses contradictory PDO mapping patterns

acolomb, 2024-07-04:
The example uses the `RemoteNode.load_configuration()` API to send `ParameterValue` settings given in the sample EDS file to the device.  This used to work in conjunction with `PdoBase.read()` to synchronize the locally cached PDO configuration and mapping info.  Calling that method (on any PDO) was therefore not really necessary, as it was implied as a final step in `load_configuration()`.

But #427 changed this and made `load_configuration()` apply the PDO configuration through calls to `PdoBase.save()` instead.  Thus the need to read it back was eliminated.

In any case, the example goes on to call `BaseNode402.setup_402_state_machine()`, which then reads back the PDO configuration again.  Here it would help to pass `read_pdos=False` since we know that the configuration was just saved, skipping the unnecessary upload.

That convenience method also checks whether needed values like the status and control words are configured in the respective PDOs.  Which should succeed because of the initialization values in the sample EDS (haven't tested).

But then the example code continues with re-mapping the PDOs, which (probably?) invalidates the `rpdo_pointers` and `tpdo_pointers` entries pointing at the previous PDO mapping's `PdoVariable` objects.  This should be done before `setup_402_state_machine()` to avoid this error.  In addition, the example first uses `node.tpdo.read()` and `node.rpdo.read()` before adjusting the mappings, which, again, is unnecessary after `load_configuration()`.

Furthermore, before the PDO re-mapping, it tries to change the `node.state` attribute, but there is no prior change of `node.nmt.state = 'OPERATIONAL'`, thus the exchanges of control and status words cannot work.  There is a fallback to SDO implemented if no PDO maps the required objects, but that is not triggered because they are in fact mapped.  The problem is that in PRE-OPERATIONAL state, the node's PDOs simply do not function.  Re-mapping them after the previous `PdoVariable` objects have been cached (see above) doesn't make this any better.

These are some observations from trying to understand the sample code, showing that it is mostly broken for actual usage and suggests wrong or inefficient patterns of the library API usage.  There may be more problems hidden, which could be revealed after testing with actual hardware.  Maybe that was done at some point, but probably not after the library API has evolved and some functionality (especially regarding profile CiA402) moved around.

We should discuss how to make this example actually useful again, or at the very minimum, document that it serves only as a demonstration of the API and will not work with real hardware unless the user takes care of the issues with their hopefully firm CANopen background themselves.

erlend-aasland, 2024-07-05:
> We should discuss how to make this example actually useful again, or at the very minimum, document that it serves only as a demonstration of the API and will not work with real hardware unless the user takes care of the issues with their hopefully firm CANopen background themselves.

FWIW, I think it would make sense to convert it to a full-fledged [tutorial](https://diataxis.fr/tutorials/), alternatively a more concise [how-to](https://diataxis.fr/how-to-guides/).

sveinse, 2024-07-06:
The way a user needs to setup any node is highly dependent on the node capabilities.
* Some node might not have any PDO configuration on startup and needs `.load_configuration()`.
* Some nodes have static PDO configuration and the local PDO configuration must be loaded from the OD (`pdo.read(from_od=True)`).
* Or perhaps the node has a variable PDO (compared to the OD) and it needs to be fetched from the node before use (`.pdo.read(from_od=False)`).
I think the examples should highlight these three use cases

erlend-aasland, 2024-07-06:
This sounds like a series of _How-to guides_ to me.

romainreignier, 2024-10-02:
I wanted to try the 402 example and faced the same issue. Do you guys have a working example for 402?

acolomb, 2024-10-02:
Sorry, nobody has taken the time to revamp this example yet.  I do not have any off-the-shelf working script for CiA 402 operation.

My advice currently is to understand the spec, figure out what configuration you actually *want to* achieve and then pick out the API calls necessary to achieve that.  It should be rather simple with the convenience methods provided, but you probably need to figure out a sensible PDO mapping first.  Or see what the device manufacturer has as a default mapping, that usually works quite well if the device advertises 402 compliance.

Of course I understand that a ready-to-run generic example will be helpful as a starting point.  Maybe you can come up with such a simple script after figuring it out for yourself?  Then we could integrate that here.  It's always better to write "beginner's" documentation after actually working with someone less experienced.  If I were to write a beginner's tutorial, I'd need to pretend not knowing what I do know, which often misses the goal.

acolomb, 2024-10-02:
Shouldn't the `Modes of operation` object be mapped to an RPDO?  Better to use numeric references here, as we don't know what your EDS files names each parameter.  If you are in fact not calling any code related to RPDO, then those will have your device's default mappings.  Check and see what those are as well.

In general, you need to call the `setup_402_state_machine()` method _after_ configuring and saving you PDO mappings.  If you've saved them via the code above (and not just store the config in the device's persistent memory), then you can pass `read_pdos=False` to that method, to skip reading back (uploading) everything you just happened to save (download).

Heideachim, 2025-06-24:
Thanks guys for your comment within this thread - without them I never would gotten my 402 running.

What helped my was checking my PDO configuration:
```
for no in range(1, 5):
    print(f"COB-ID of {node.tpdo[no].name}: 0x{node.tpdo[no].cob_id:04X}")
    for var in node.tpdo[no].map:
        print(f"   0x{var.index:04X} subindex {var.subindex}: {var.name}")
    print(f"COB-ID of {node.rpdo[no].name}: 0x{node.rpdo[no].cob_id:04X}")
    for var in node.rpdo[no].map:
        print(f"   0x{var.index:04X} subindex {var.subindex}: {var.name}")
```

But then, also check if the library has digested all of your efforts: 
```
for key, value in node.tpdo_pointers.items():
    print(f"TxPDO: 0x{key:04X}: {value}")    
for key, value in node.rpdo_pointers.items():
    print(f"RxPDO: 0x{key:04X}: {value}")
```
The output should contain the same objects as your PDO listing!

As stated above, `setup_402_state_machine()` must be called **after** you made you configuration, otherwise it will fail.