canopen-python/canopen #502: Type annotation of Mapping in Python 3.8

sveinse, 2024-07-08:
This is a discussion related to typing, #358. 

canopen use `Mapping` and `MutableMapping` for 10 classes. Due to limitations in Python 3.8, they cannot be annotated and the following will not work

```py
class Network(MutableMapping[int, Union[RemoteNode, LocalNode]]):
```

Python 3.8 gives the error: `TypeError: 'ABCMeta' object is not subscriptable`. The effect is that a code base supporting 3.8 cannot use type hints on `Mapping`. These classes are very central to the function of canopen, so I think we'd miss out on significant type annotation aids if we don't get to use it.

There are a few options available to us:

1. Bump the minimum Python version to 3.9
2. Don't use type annotations on any of the `Mapping` and `MutableMapping` classes because we need Python 3.8 compatibility
3. Use a run-time hack that conditionally adds the type annotations.  Below example is inspired from [this example](https://github.com/squidfunk/mkdocs-material/issues/5916#issuecomment-1697502342)
4. Some way to annotate these classes that works with >=3.8 that I don't know about

```python
# Example to annotate MutableMapping that makes it run on Python 3.8
import sys
if sys.version_info >= (3, 9):
    TMutableMapping = MutableMapping[int, Union[RemoteNode, LocalNode]]
else:
    TMutableMapping = MutableMapping

class Network(TMutableMapping):
    """Representation of one CAN bus containing one or more nodes."""
````

erlend-aasland, 2024-07-08:
FTR, Python 3.8 goes EOL in October 2024.

acolomb, 2024-07-08:
I'd avoid jumping to a 3.9 requirement soon.  It's very common to work on devices with older distributions when hardware interaction is concerned, so waiting some time until after Python 3.8 has EOLed seems sensible.

Do we really have to introduce a subindexed `MutableMapping` type here?  The results should be very useful already with the generic type - being able to check methods and their signatures at least.  And for the other classes, it will get messy trying to list all kinds of possible entry objects.  We could just stick with the status quo and revisit once there is a stronger  reason (and better time) to bump our Python requirement.

For the typing, I absolutely prefer the `str | int` syntax instead of `Union[str, int]` for example.  But as we're trying hard to be backwards-compatible, we'll just need to live with the older syntax for a while longer.

sveinse, 2025-06-22:
This is related to #585 , where python-can has bumped minimum version to 3.9. I assume that we want to make the minimal version to 3.9 in the 3.0.0 release, so I'm labelling this issue as such.

acolomb, 2025-08-05:
Just had a look at this again, but subscripting `collections.abc.MutableMapping` is apparently still not supported.  If we `from typing import MutableMapping` instead, we lose out on the added functionality from that base class though, don't we?

What exactly do you propose to change here, now that we have the power of Python 3.9 annotations at hand?