canopen-python/canopen #514: Improve test coverage

erlend-aasland, 2024-07-10:
I'd like to raise the test suite cover more of the code base, as a part of the preparations for #358.

Currently, `coverage.py` reports 73% code coverage when run locally:

```
Name                                    Stmts   Miss  Cover
-----------------------------------------------------------
canopen/__init__.py                        12      2    83%
canopen/_version.py                        11      2    82%
canopen/emcy.py                            71     19    73%
canopen/lss.py                            181    106    41%
canopen/network.py                        165     21    87%
canopen/nmt.py                            133      2    98%
canopen/node/__init__.py                    2      0   100%
canopen/node/base.py                        9      0   100%
canopen/node/local.py                      81     10    88%
canopen/node/remote.py                     81     37    54%
canopen/objectdictionary/__init__.py      306     37    88%
canopen/objectdictionary/datatypes.py      82      8    90%
canopen/objectdictionary/eds.py           349     38    89%
canopen/objectdictionary/epf.py            81     81     0%
canopen/pdo/__init__.py                    36      8    78%
canopen/pdo/base.py                       374    108    71%
canopen/profiles/__init__.py                0      0   100%
canopen/profiles/p402.py                  277    176    36%
canopen/sdo/__init__.py                     5      0   100%
canopen/sdo/base.py                        93     20    78%
canopen/sdo/client.py                     461     92    80%
canopen/sdo/constants.py                   27      0   100%
canopen/sdo/exceptions.py                  14      5    64%
canopen/sdo/server.py                     127      9    93%
canopen/sync.py                            19      7    63%
canopen/timestamp.py                       15      0   100%
canopen/utils.py                           13      1    92%
canopen/variable.py                        98     42    57%
-----------------------------------------------------------
TOTAL                                    3123    831    73%
```

73% is very promising; getting up to 80% is the next milestone.

In #504, I raised the coverage of `canopen/nmt.py` to 98%. I'm going to continue with `canopen/network.py` (anticipate PR in a day or two). Feel free to join!

I'll update this list as progress is made:

- [ ] `canopen/emcy.py`
- [ ] `canopen/lss.py`
- [ ] `canopen/network.py`
- [x] `canopen/nmt.py`: #504
- [x] `canopen/node/base.py`
- [ ] `canopen/node/local.py`
- [ ] `canopen/node/remote.py`
- [ ] `canopen/objectdictionary/__init__.py`
- [ ] `canopen/objectdictionary/datatypes.py`
- [ ] `canopen/objectdictionary/eds.py`
- [ ] `canopen/objectdictionary/epf.py`
- [ ] `canopen/pdo/__init__.py`
- [ ] `canopen/pdo/base.py`
- [ ] `canopen/profiles/p402.py`
- [ ] `canopen/sdo/base.py`
- [ ] `canopen/sdo/client.py`
- [x] `canopen/sdo/constants.py`
- [ ] `canopen/sdo/exceptions.py`
- [ ] `canopen/sdo/server.py`
- [ ] `canopen/sync.py`
- [x] `canopen/timestamp.py`
- [ ] `canopen/utils.py`
- [ ] `canopen/variable.py`

acolomb, 2025-07-21:
@yuvraj-kolkar17 you're very welcome to help in this area.  I haven't heard from @erlend-aasland for a few months in this project.  The issue is still valid though and corresponding PRs will be reviewed.

Just please make sure to keep them not too large (i.e. one module at a time roughly) and do not mix with functional changes.  Even if your added tests uncover a bug, that should be addressed separately.

yuvraj-kolkar17, 2025-07-21:
@acolomb thanks for the guidance. 
I’ve added a complete unit test suite for the canopen/emcy.py module, which improves the test coverage from 73% to 99%.

The PR is here: #604 
Would appreciate it if you could take a look and provide feedback whenever convenient. 

Thanks!