canopen-python/canopen #520: Object Dictionary does not get updated

Phillip0407, 2024-07-12:
I have a real CAN-network with one device connected. This device tries to interact with another node that I want to simulate by a LocalNode object, which workds fine. I created an object dictionary with one ODVariable with the particular index and subindex on which the real device tries to perform an SDO Download (e.g. 0x1F51, 0x01) and assigned it to the LocalNode object.
The SDO Download completes successfully what I can observe via candump. However, when I try to access the value via the object dictionary from the local node, I can't see the updated value but the default one from the initialization. The only way I can access the updated value is via "node.sdo[0x1F51].raw", which is an SdoVariable and not the original ODVariable. At this point, I can not provide any subindex and when a second ODVariable got assigned before with the same index but different subindex, I only can check the updated value from the most recently added Variable.

I don't know if I understand wrong the concept of the different variable classes, but my guess was that the object dictionary of the LocalNode gets updated when another Node performs SDO Downloads.

If not, how can I access an Variable with index and subindex?

acolomb, 2024-08-17:
You're right in your expectation about the OD getting updated.  That's what a `LocalNode` should be doing through its `SdoServer` member.

Now I think you might be looking at the wrong thing and therefore coming to wrong conclusions.  The `ODVariable` holds all the metadata about the object, and especially the actual value for a local SDO node.  The `SdoVariable` is pretty much only an adapter to make SDO reading and writing easier.  It points back to its associated `ODVariable`.

Now accessing local data through SDO is not wrong, as you might have thought.  It basically just accesses the current cached data in the OD.  There is no network overhead involved when calling this for a `LocalNode`.  But of course you can also just look up the `ODVariable` object and use that directly the data should be identical.

> If not, how can I access an Variable with index and subindex?

There is a `get_variable()` method on both the object dictionary and the SDO server class.  That's how you easily find the right variable.  But beware of the second argument's default value: if not specified, it will give you subindex 0 of any record or array.

elmout31, 2025-01-28:
tag version 2.1.0. 
Hello, just to add more details as I encounter  exactly the same situation : Try to simulate a node and access local object dictionnary. 

1) Usage of create_node to load  an hardware EDS.  My default eds file value for object 0x1006 ('Communication cycle period') is 50000 
2) From external equipement, i am using SDO to write into object 0x1006 (.phys) 40000. (This object is the the sync period).
3) On the local node, i have tried to get the 0x1006 object value  by using this methods:
* node.object_dictionary[0x1006]
* node.object_dictionary['Communication cycle period']
* node.object_dictionary['Communication cycle period']|0]
* node.object_dictionary[0x1006][0]
* node.object_dictionary.getVariable(0x1006) 
* node.object_dictionary.getVariable(0x1006,0) 

All above method return an object which is the copy of the eds file and not the updated values. 
The object properties are as follow:
 * values = None 
 * default = 50000
 * default_raw = 50000

I found correct data updated inside another object named node.datastore which is access by sdo methods.
It confirms that the only thing that is working is passing by SDO read methods to access the object dictionnary even in Local node.
eg:
syncCycle = node.sdo[0x1006].phys

What is missleading, is that inside the documentation https://canopen.readthedocs.io/en/latest/od.html, we may read: 
```
device_name_obj = node.object_dictionary['ManufacturerDeviceName']
vendor_id_obj = node.object_dictionary[0x1018][1]
actual_speed = node.object_dictionary['ApplicationStatus.ActualSpeed']
command_all = node.object_dictionary['ApplicationCommands.CommandAll']
```
**This code  does not work for local node.**

The documentation should be updated.

Regards
Julien