canopen-python/canopen #548: IntegerN class and UnsignedN class issue

hhhaaannnsss, 2024-11-13:
### issue

Starting python and running "import canopen" leads to the following error:

```
$ python 
Python 3.12.0 (v3.12.0:0fb18b02c8, Oct  2 2023, 09:45:56) [Clang 13.0.0 (clang-1300.0.29.30)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import canopen
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Users/hans/gitlab/canopen/canopen/__init__.py", line 1, in <module>
    from canopen.network import Network, NodeScanner
  File "/Users/hans/gitlab/canopen/canopen/network.py", line 13, in <module>
    from canopen.node import LocalNode, RemoteNode
  File "/Users/hans/gitlab/canopen/canopen/node/__init__.py", line 1, in <module>
    from canopen.node.local import LocalNode
  File "/Users/hans/gitlab/canopen/canopen/node/local.py", line 7, in <module>
    from canopen import objectdictionary
  File "/Users/hans/gitlab/canopen/canopen/objectdictionary/__init__.py", line 309, in <module>
    class ODVariable:
  File "/Users/hans/gitlab/canopen/canopen/objectdictionary/__init__.py", line 318, in ODVariable
    INTEGER24: IntegerN(24),
               ^^^^^^^^^^^^
TypeError: Struct() argument 1 must be a str or bytes object, not int
>>> 
```

### root cause
Changing the "repo_root"/canopen/objectdictionary/__init__.py as below fixes the problem
[image]

Even though the child class constructor takes in an int and calls the superclass constructor with a string, my systems leads to above error. Not exactly sure where it goes wrong.

[image]

[image]

### context
Python 3.12.0
```
pip freeze                                                             
...
-e git+https://github.com/christiansandberg/canopen.git@ffbd10f44da89e407622c82b8587fcc64e3f2f49#egg=canopen
certifi==2023.7.22
python-can==4.4.2
....
```

acolomb, 2024-11-13:
Please have a look at issue #470, where the same problem was previously reported.  Apparently it is caused by some oddness in Python 3.12, which was fixed in a later point release.  Version 3.12.4 was reported to not show the issue.

erlend-aasland, 2024-11-13:
> Version 3.12.4 was reported to not show the issue.

Tried to replicate this using 3.12.7 at canopen commit ffbd10f44da89e407622c82b8587fcc64e3f2f49, but failed to do so:

```python
$ python 
Python 3.12.7 (v3.12.7:0b05ead877f, Sep 30 2024, 23:18:00) [Clang 13.0.0 (clang-1300.0.29.30)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import canopen
>>> exit()
```

hhhaaannnsss, 2024-11-13:
Thank you and apologies for re-reporting

closed 2024-11-13 by hhhaaannnsss