canopen-python/canopen #549: Error BUS HEAVY BUS LIGHT

SyedArif245, 2024-11-21:
We are using D 20 9 pressure Sensor, but sensor is not communicating(not sending power up message)

acolomb, 2024-11-22:
What kind of error message do you get and where do you see it?

The "BUS HEAVY" and "BUS LIGHT" labels I recognize from Peak's PCAN-View software. Is that where you see them? In that case, it indicates a physical problem with your CAN bus.

* Cabling (conductors swapped somewhere)
* Terminating resistors (need 120 Ohm on each end)
* Mismatched bit rate

SyedArif245, 2024-11-23:
yes BUSHEAVY and BUS LIGHT I have seen that in PCAN software
before at one end we connected terminating _resistor._
now i connected on the other end as well
but still sensor is not communicating

Baudrate was configured to 500kbps Same given in PCAN as well.
""Cabling (conductors swapped somewhere)""
the above point i am unable to understand

sveinse, 2024-11-24:
This doesn't sound like a canopen issue which is the topic for this project. It sounds like an issue with that module or the CAN bus as earlier stated. A tip could be to start with a CAN-bus configuration that's known to work and change one thing at a time. E.-g. so you have another CAN module you're able to connect to and communicate to using the PCAN adapter without connecting the D20 sensor? If this works without the sensor, but fails to work when the D20 sensor is connected, you know there is something related to the sensor that's not working.

Best of luck.

SyedArif245, 2024-11-26:
Basically it is a CANopen pressure sensor 
it is an external pressure sensor.
I am facing issue with this,
Can you clarify that pressure can be send with out giving the power supply to sensor??

SyedArif245, 2024-11-26:
If sensor is not connected to power supply but the still air pressure sent to sensor
the i this case is there any chance of damaging internal Components?

acolomb, 2024-11-26:
> Can you clarify that pressure can be send with out giving the power supply to sensor??

I think @sveinse misread the question.  *WithOUT* supplying power to the sensor, it can never function as a CANopen device.  CANopen is an active communication protocol and it needs electrical power on each device.

SyedArif245, 2024-11-26:
> I think @sveinse misread the question. _WithOUT_ supplying power to the sensor, it can never function as a CANopen device. CANopen is an active communication protocol and it needs electrical power [...]

yes i understood your point
my question was if i pressurised the sensor with out giving power supply,
the is there any chance of internal damage pf sensor

SyedArif245, 2024-11-26:
> > the i this case is there any chance of damaging internal Components?
> 
> That's a question you will need to ask your sensor supplier. I don't know of any pressure sensors that need electrical power to protect themselves from damage through applied gas pressure.
> 
> But really, this all has nothing to do with CANopen or with this Python library. You should learn the basics about your sensing business and some electronics / CAN basics before trying to use this library.

 
Thank you i wil check