canopen-python/canopen #554: node.status not matching contents of node.sdo["Statusword"].raw

bentopologic, 2025-01-07:
As the title would suggest there is a mismatch between the Statusword read from the motor, and the node.status generated from the BaseNode402 state machine. Which is then causing a `RuntimeError: Timeout waiting for updated statusword` at line 52 in my script. 

The log data that shows the discrepancy is on line 9 of the shortened log file. (A log statement generated by line 51 of my script). 

I found Issue #152 and followed along with what had been suggested there, but still couldn't get it to cooperate. I am fairly confident it doesn't have anything to do with the EDS file because I can still use the object dictionary to read the statusword value with `node.sdo["Statusword"].raw` . Any thoughts?

**THE SCRIPT**
```
import canopen
from LM1483_11Motor import LM1483_11
import time
import logging
import datetime
from pathlib import Path

log_name = Path.cwd()/'logs'/datetime.datetime.now().strftime('breadboard_log_%Y-%m-%d-%H-%M-%S.log')
logging.basicConfig(level=logging.DEBUG, filename=log_name, format='%(asctime)s - %(name)s - %(levelname)s - %(message)s')

eds_file = "C:/Users/benveghte/Documents/Gitlab/breadboard-hardware/M5005 S CO LM1493_XX_11.eds"

network = canopen.Network()
network.connect(interface='pcan', channel = 'PCAN_USBBUS1', bitrate=1000000)
network.check()

motor = LM1483_11(canopen.BaseNode402(33, eds_file))

network.add_node(motor.node)

# motor.begin_communication()
motor.node.nmt.state = 'RESET COMMUNICATION'
motor.node.nmt.wait_for_bootup(10)
motor.node.nmt.state = "OPERATIONAL"
motor.node.nmt.wait_for_heartbeat()
logging.info(f"Node {motor.node.id} connected and has a heartbeat")

#Make sure that there are no errors in the controller
#Index 0x1003 is the pre-defined errors
num_errors = motor.node.sdo[0x1003][0].raw
if num_errors != 0:
    for i in range(num_errors):
        curr_error = motor.node.sdo[0x1003][i+1].raw
        hrs = motor.error_parser(curr_error)[3]
        raise Exception(f"Node {motor.node.id} Error: {hrs}")

time.sleep(0.5)

#Make sure the locally stored PDOs align with those on the motor controller
motor.node.rpdo.read()
motor.node.tpdo.read()

#Begin the 402 state machine
motor.node.setup_402_state_machine()

network.sync.start(0.1)
time.sleep(0.5)

motor.set_profile_position_mode()
motor.set_profile_velocity(3000)
logging.debug(f"Curr State: {motor.node.state} Curr Statusword: {motor.node.sdo["Statusword"].raw:_b}")
motor.node.state = "SWITCHED ON"
time.sleep(0.5)
print(motor.node.state)
```

**The log file segment**
```
2025-01-07 17:00:10,413 - canopen.sdo.client - DEBUG - Reading 0x6041:00 from node 33
2025-01-07 17:00:10,413 - can.pcan - DEBUG - Data: bytearray(b'@A`\x00\x00\x00\x00\x00')
2025-01-07 17:00:10,414 - can.pcan - DEBUG - Type: <class 'bytearray'>
2025-01-07 17:00:10,414 - canopen.variable - DEBUG - Value of 'Position actual value' (0x6064:00) is -3200
2025-01-07 17:00:10,414 - canopen.variable - DEBUG - Value of 'Position demand value' (0x6062:00) is 0
2025-01-07 17:00:10,414 - canopen.variable - DEBUG - Value of 'Velocity actual value' (0x606C:00) is 0
2025-01-07 17:00:10,414 - canopen.variable - DEBUG - Value of 'Velocity demand value' (0x606B:00) is 0
2025-01-07 17:00:10,416 - canopen.variable - DEBUG - Value of 'Statusword' (0x6041:00) is 64
2025-01-07 17:00:10,416 - root - DEBUG - Curr State: NOT READY TO SWITCH ON Curr Statusword: 100_0000
2025-01-07 17:00:10,416 - canopen.variable - DEBUG - Writing 'Controlword' (0x6040:00) = 0
2025-01-07 17:00:10,416 - canopen.pdo.base - DEBUG - Updating Controlword to b'0000' in RxPDO1_node33
2025-01-07 17:00:10,416 - can.pcan - DEBUG - Data: bytearray(b'\x00\x00\x01')
2025-01-07 17:00:10,416 - can.pcan - DEBUG - Type: <class 'bytearray'>
2025-01-07 17:00:10,511 - can.pcan - DEBUG - Data: bytearray(b'')
2025-01-07 17:00:10,512 - can.pcan - DEBUG - Type: <class 'bytearray'>
2025-01-07 17:00:10,514 - canopen.variable - DEBUG - Value of 'Position actual value' (0x6064:00) is -3200
2025-01-07 17:00:10,514 - canopen.variable - DEBUG - Value of 'Position demand value' (0x6062:00) is 0
2025-01-07 17:00:10,514 - canopen.variable - DEBUG - Value of 'Velocity actual value' (0x606C:00) is 0
2025-01-07 17:00:10,514 - canopen.variable - DEBUG - Value of 'Velocity demand value' (0x606B:00) is 0
2025-01-07 17:00:10,611 - can.pcan - DEBUG - Data: bytearray(b'')
2025-01-07 17:00:10,611 - can.pcan - DEBUG - Type: <class 'bytearray'>
2025-01-07 17:00:10,614 - canopen.variable - DEBUG - Value of 'Position actual value' (0x6064:00) is -3203
2025-01-07 17:00:10,614 - canopen.variable - DEBUG - Value of 'Position demand value' (0x6062:00) is 0
2025-01-07 17:00:10,614 - canopen.variable - DEBUG - Value of 'Velocity actual value' (0x606C:00) is 0
2025-01-07 17:00:10,614 - canopen.variable - DEBUG - Value of 'Velocity demand value' (0x606B:00) is 0
```

acolomb, 2025-01-13:
Your script does not contain the mapping of your motor's TPDO.  Are you sure the Statusword (0x6041) is actually mapped into the PDO?  Otherwise it will fall back to reading it via SDO.  The log output from the `*pdo.read()` methods would be interesting to verify that.

What's also strange about your script: You're first setting the drive to OPERATIONAL, then going on with the DS402 and PDO setup.  Usually you should do all of that configuration stuff in the PRE-OPERATIONAL state, from which you can also change PDO mappings.

bentopologic, 2025-03-12:
I think there was something weird happening on my end, never figured out what exactly was causing it but, I stopped getting the error

closed 2025-03-12 by bentopologic