canopen-python/canopen #558: LocalNode and TPDO

KaSroka, 2025-02-13:
What is the correct usage of TPDO with local nodes? I'm writing code for slave device. I'm adding new variable to TPDO with
```python
fcc = node.tpdo[2].add_variable("FullCharge Capacity")
```
Unfortunately I can see that the value of TPDO variable and SDO variable are disconnected. So that
```python
node.sdo["FullCharge Capacity"].phys != fcc.phys
```

Is there something I'm missing or this is not implemented as of now?

acolomb, 2025-02-16:
This is simply not implemented.  The library in its current state is more of a toolbox to handle the CANopen protocol layer.  It is not a complete communication logic implementation for a local device.  You will need to listen to various objects and let your notifier callbacks handle the necessary adjustments to a common data store.  Whatever your local node implementation does, it's certainly not limited to keeping TPDO and SDO response data synchronized.  Because the other requirements of a device's logic are usually much more complex and diverse, the library doesn't try to provide a dumb implementation that just keeps a common Object Dictionary data model.

The other reason is that the local node functionality was tacked on later, and is definitely less mature / complete than the remote node features.  Lack of developers' time and interest basically.  If someone were to contribute a sensible default implementation to make local nodes behave closer to standard expectations out of the box, adding it to the library could very well be considered.

KaSroka, 2025-02-16:
Understood, thank you.

closed 2025-02-16 by KaSroka