canopen-python/canopen #561: Transmit PDO when nmt.state is not OPERATIONAL

PaulFirs, 2025-02-24:
Currently, PDOs are being transmitted even when the NMT state is not OPERATIONAL.

acolomb, 2025-02-24:
Are you talking about a `LocalNode` instance?  If so, you have to understand that this library does not provide the complete device logic for a CANopen device, but merely the building blocks to integrate the protocol with your application.  That could be extended, but the requirements are not the same for all apps and users may want to couple further logic with the NMT state switching.

If that is your underlying problem, then you should subscribe to the NMT state change commands and enable or disable your PDO transmissions in your handler function.

Otherwise, please elaborate on your use-case and what you expected to happen.

PaulFirs, 2025-02-24:
@acolomb Yes, I am using a LocalNode. I have added a callback function for the SYNC event network.subscribe(0x080, sync_pdo_callback). When a node receives SYNC, I check the PDOs and, if the PDO type is for the SYNC event, I transmit the PDO using node.tpdo[1].transmit(). I expected that there would be a check for the NMT state in the transmit function to ensure the node is in the OPERATIONAL state, as in CANopen, PDOs should only be transmitted in the OPERATIONAL mode. However, regardless of the NMT state, the PDO is being sent.

acolomb, 2025-02-24:
Well then you need to extend your check to also take the local NMT state into account.  If you tell the library to transmit a PDO, it will do so.  That is a very basic function, not trying to account for all of the CANopen communication logic in each API call.