canopen-python/canopen #563: TIME producer does not correct provided timestamp

mrrosen, 2025-02-25:
The documentation on `transmit` suggests the provided timestamp should be a UNIX timestamp, so have Epoch Jan 1 1970 UTC. However, the function does not correctly offset the provided timestamp as it does with the one produced internally if no timestamp is provided.

Either the documentation should be fixed to say the provided timestamp should use the CANopen epoch or the like correcting the timestamp should be fixed to respect Python operator presence:

```
delta = (timestamp or time.time()) - OFFSET
```

sveinse, 2025-04-26:
Unless I'm completely mistaken, the order of precedence is preserved in both variants:

```py
# These are equivalent
delta = timestamp or time.time() - OFFSET
delta = (timestamp or time.time()) - OFFSET
```

When the provided `timestamp` is based on epoch 1970, `OFFSET` needs to be subtracted to get it to canopen January 1, 1984 epoch.

mrrosen, 2025-04-26:
> Unless I'm completely mistaken, the order of precedence is preserved in both variants:
>
> ```py
> # These are equivalent
> delta = timestamp or time.time() - OFFSET
> delta = (timestamp or time.time()) - [...]

These expressions are not the same, `-` comes before `or` in operator precedence: https://docs.python.org/3/reference/expressions.html#operator-precedence

Correct that `OFFSET` should be subtracted from either `timestamp` argument or `time.time()` for a time in CANopen epoch, but the current expression doesn't do that.

sveinse, 2025-04-26:
Yeah, I see it now

```
>>> 0 or 1 - 5
-4
>>> (0 or 1) - 5
-4
>>> None or 1 - 5
-4
>>> (None or 1) - 5
-4
>>> 1 or 2 - 5
1
>>> (1 or 2) - 5
-4
```

PR #576 (2025-06-11): Correctly apply offset to custom timestamp value (fixes #563)
Fix issue when providing a custom timestamp. Updated the test case to test provided and not provided timestamp.

Fixes #563