canopen-python/canopen #568: Add support for parsing XDC (XML Device Configuration) files

juliusz-t, 2025-04-22:
It would be nice to add support for the [CiA 311 XML device description](https://www.can-cia.org/can-knowledge/cia-311-xml-schema-definition). Currently, library supports parsing EDS, DCF, and EPF files via `canopen.objectdictionary.import_od()` and `canopen.network.Network.add_node()`. Attempting to load an `.xdc` file results in a `NotImplementedError`.

The XML device description is mandatory for CANopen FD devices. Unfortunately the format specification is hidden behind the member wall. :/

acolomb, 2025-04-22:
Are there any other open source implementations that already implement the format?  I don't have access to the (non-public) specs either.  The feature would definitely be welcome and fit in nicely (with an adequate implementation).  But I don't see a way to implement it without having some kind of format description available.  Whoever gets access to it is welcome to contribute an implementation here.

rusoku2, 2025-04-24:
I checked and found that this document is very old. But maybe there is no newer version.

Draft Specification Proposal
CiA 311
Version 1.1.0
August 10, 2011
CANopen XML schema definition

juliusz-t, 2025-04-24:
There is a newer document [CiA 1311](https://www.can-cia.org/can-knowledge/cia-1311-canopen-fd-electronic-device-description), relating directly to the FD standard. I think both are still relevant though.

rusoku2, 2025-04-24:
Ok thanks.I found.

© CAN in Automation (CiA) e. V.
Draft Specification Proposal
CiA 1311-1
Version 1.0.1
August 15, 2022
CANopen FD electronic device descriptions
Part 1: XML schema definitions for CiA 1301