canopen-python/canopen #584: Callback on a periodic RPDO transmit?

BCMicBem, 2025-06-04:
Hello,
I have an entry in the OD that needs to be incremented with each RPDO1 transmission ("Lifetime"). The entry is mapped into the said RPDO1.
How can I get notified that the RPDO1 was transmitted by the task started with
`node.rpdo[1].start(0.200)`
for instance?
The callback
`node.rpdo[1].add_callback(OnRPDO1)`
does not work as neither
`node.network.subscribe(node.rpdo[1].cob_id, OnRPDO1)`
does.
Any hints?

Thank you very much for your excellent work!

setteZ, 2025-09-01:
@BCMicBem I am having the same issue, and looking at the documentation of python-can I found out that [send_periodic()](https://python-can.readthedocs.io/en/stable/bus.html#can.BusABC.send_periodic) has this feature. I am working to it [here](https://github.com/setteZ/canopen/tree/periodic_rpdo_modifier_callback)

@acolomb do you think it is worth integrating this approach with a PR?

acolomb, 2025-09-05:
Frankly I think that's not good software architecture to set up a timer for transmission, then invoke a user-provided callback before it actually does transmit.  It gives you no guarantees on timing, because the callback could take arbitrarily long to complete.

It's not like the timer thread started by `pdo.start()` is anything special or complicated.  If you want it to do more, then just start your own thread which pulls the updated value and then calls `pdo.transmit()`.  It doesn't *need* to be handled by `send_periodic()` at all, unless you need the cyclic transmission implemented on the CAN controller level for some interface types.

So I guess a PR with those changes would have small chances.  This is a library providing tools to interact with a CANopen network.  Not an application framework that tries to manage all your application logic with callback hooks for each possible interaction point.  In my experience, it makes more sense to manage scheduling and timing in your application code, where it will usually interact with other things than just the CANopen communication.  Use whatever source of the changing RPDO data as the trigger to update it (push model).  If you don't care about the overall timing, the built-in periodic transmission task can still be used and run asynchronously, using whatever value the RPDO content currently holds at the time of scheduled transmission.  If you really do need to update data on an exact schedule (pull model), then implement that and just add a `pdo.transmit()` call to it.  As your application grows, that ensures you have a consistent overall transmission cycle instead of various independent tasks which can easily become unsynced because of timing jitter and drift.